particle-iot/shields #51: Threads in OpenOCD

hackrid, 2016-03-03:
Hi,

i recently found your wiki entry

m-mcgowan, 2016-03-03:
I'm using openOCD with FreeRTOS. I launch using this command line
```
openocd -f interface/ftdi/particle-ftdi.cfg -f target/stm32f2x.cfg -c "stm32f2x.cpu configure -rtos auto" &
```

hackrid, 2016-03-04:
thanks for your attention on this.

I got it working with the following command line:
`-f board/stm32429i_eval_stlink.cfg -c “$_TARGETNAME configure -rtos auto”`

how is your impression using OpenOCD with Thread Awareness? I experience a horrible single stepping speed since the back traces of the individual threads are built up every time....

m-mcgowan, 2016-03-15:
It's not as fast as debugging a local executable but it's workable - single step times no more than 1 second, I'd guess, probably less.

closed 2016-04-21 by hackrid