particle-iot/shields #65: Add "which pins are used" designators to Asset Tracker

jenesaisdiq, 2017-02-12:
Asset Tracker v2 design doesn't indicate which pins are being used by the shield on the silkscreen. This makes it difficult to know, at a glance, which pins are available for other uses.

MrMarkA, 2017-03-17:
I am pretty sure I have the V2 board (v002 near the CR1220 battery holder) The pins used on the shield by the Electron are denoted on the silkscreen of the shield by an open circle next to the pin identifier.  (i.e. oRX, oTX, oA5, oA4, oA3, oA2, and on the other side D6o)  This matches the pins used from the schematic.

jenesaisdiq, 2017-03-19:
Sorry @MrMarkA, that's a comment on an unreleased revision of the hardware for @mohitbhoite to address. Brings up a good point, though- we should really rethink how those on-PCB version numbers work as well. Is the one I'm referring to going to be V201, following semver?